lvgl/lv_lib_png #2: memory allocation failed

simonchen007, 2020-05-18:
hello,

I'm using esp32 + lvgl + lv_lib_lodepng,
but when I use a png more 20K , it would shows "memory allocation failed", where can I change for making the memory allocation bigger ?

embeddedt, 2020-05-18:
You probably have to increase `LV_MEM_SIZE` (or the size of whatever heap you use).

simonchen007, 2020-05-18:
I change it as   (64U * 1024U),still the same. however, I change it as 128U * 1024U , I would goes to compile error.

kisvegabor, 2020-05-19:
lodepng uses normal `malloc`. So you are simply out of heap on ESP.

simonchen007, 2020-05-19:
so, are you meaning the problem is because the esp32 having not enough money ?
I tested over 50 times last night, if the PNG is over 5kb, I would get this error.
however, if I use the normal way(make the png as c-array first), and do not use lodepng, 100kb png file is also works.

kisvegabor, 2020-05-19:
> so, are you meaning the problem is because the esp32 having not enough money ?

I don't know how much **money** ESP has but it might be out of **memory**. :slightly_smiling_face: : 

The resolution of the PNG matters. During decoding the PNG image a `hor. res x ver.res. x color size` array is allocated into RAM. E.g. for 200 px x 200 px x 2 bytes = ~80kB

simonchen007, 2020-05-19:
esp32 having 320K memory.
And I read the code of lodepng, "memory allocation failed" is because case 83 . if esp32 out of memory, I would not send any error, it would just go reboot.

kisvegabor, 2020-05-19:
What is the resolution of the image?

simonchen007, 2020-05-19:
320*480

kisvegabor, 2020-05-19:
It means the decompressed image is 320*240*2 = 300kB. If it uses 32-bit color format internally (I'm not sure now) than its 600 kB. Anyway, 320kB is not enough if either case.

simonchen007, 2020-05-19:
but why if I use the normal way(make the png as c-array first), and do not use lodepng, even 100kb png file is works ?

embeddedt, 2020-05-19:
The PNG doesn't need to be decoded into RAM first with that method. It can be drawn directly from flash to the display.

rambo-panda, 2023-02-16:
I also encountered the problem under lvgl 8.3. But I found two ways to solve it

1. `#LV_MEM_CUSTOM 1` use the `malloc` and `free` `realloc`
2. convert to *.bin file

waiting for the real solution 😄

kisvegabor, 2023-02-16:
You can increase `LV_MEM_SIZE` to add more memory for LVGL.

rambo-panda, 2023-02-17:
Yes I tried, but it still reports an error。
So I personally think it's possible that lvgl's heap management may not be able to allocate large chunks of memory due to memory fragmentation in some cases

rambo-panda, 2023-02-17:
```diff
 #if LV_MEM_CUSTOM == 0
     /*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
-    #define LV_MEM_SIZE (48U * 1024U)          /*[bytes]*/
+    #define LV_MEM_SIZE (148U * 1024U)          /*[bytes]*/
```

```log
Error]	(0.105, +105)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Error]	(0.108, +3)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Warn]	(0.108, +0)	 decoder_open: error 83: memory allocation failed
 	(in lv_png.c line #160)
[Warn]	(0.108, +0)	 _lv_img_cache_open: Image draw cannot open the image resource 	(in lv_img_cache.c line #125)
[Warn]	(0.108, +0)	 lv_draw_img: Image draw error 	(in lv_draw_img.c line #83)
[Error]	(0.109, +1)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Error]	(0.112, +3)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Warn]	(0.112, +0)	 decoder_open: error 83: memory allocation failed
 	(in lv_png.c line #160)
[Warn]	(0.112, +0)	 _lv_img_cache_open: Image draw cannot open the image resource 	(in lv_img_cache.c line #125)
[Warn]	(0.112, +0)	 lv_draw_img: Image draw error 	(in lv_draw_img.c line #83)
[Error]	(0.112, +0)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Error]	(0.115, +3)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Warn]	(0.115, +0)	 decoder_open: error 83: memory allocation failed
 	(in lv_png.c line #160)
[Warn]	(0.115, +0)	 _lv_img_cache_open: Image draw cannot open the image resource 	(in lv_img_cache.c line #125)
[Warn]	(0.115, +0)	 lv_draw_img: Image draw error 	(in lv_draw_img.c line #83)
[Error]	(0.116, +1)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Error]	(0.119, +3)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Warn]	(0.119, +0)	 decoder_open: error 83: memory allocation failed
 	(in lv_png.c line #160)
[Warn]	(0.119, +0)	 _lv_img_cache_open: Image draw cannot open the image resource 	(in lv_img_cache.c line #125)
[Warn]	(0.119, +0)	 lv_draw_img: Image draw error 	(in lv_draw_img.c line #83)
[Error]	(0.119, +0)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Error]	(0.122, +3)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Warn]	(0.122, +0)	 decoder_open: error 83: memory allocation failed
 	(in lv_png.c line #160)
[Warn]	(0.122, +0)	 _lv_img_cache_open: Image draw cannot open the image resource 	(in lv_img_cache.c line #125)
[Warn]	(0.122, +0)	 lv_draw_img: Image draw error 	(in lv_draw_img.c line #83)
[Error]	(0.123, +1)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Error]	(0.126, +3)	 lv_mem_realloc: couldn't allocate memory 	(in lv_mem.c line #211)
[Warn]	(0.126, +0)	 decoder_open: error 83: memory allocation failed
 	(in lv_png.c line #160)
```

rambo-panda, 2023-02-17:
@kisvegabor 

apologize to you. it's my mistake.

 I just committed git and found that the changelog had a change that I had debugged. 
It affected the `lvgl` allocation