lvgl/lv_lib_png #3: How to enable key color when displaying PNG

zq190kg, 2020-09-02:
[image]

Like this effect, only the main body is displayed，Thank you for your reply

zq190kg, 2020-09-02:
#define LV_COLOR_SCREEN_TRANSP    1
#define LV_COLOR_TRANSP    LV_COLOR_LIME         /*LV_COLOR_LIME: pure green*/

I defined the two line

zq190kg, 2020-09-02:
lvgl  v6.0

#define LV_COLOR_SCREEN_TRANSP 1
#define LV_COLOR_TRANSP LV_COLOR_LIME /LV_COLOR_LIME: pure green/
LV_IMG_DECLARE(png_decoder_test);
    lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(img, &png_decoder_test);

Like the effect below
[image]

embeddedt, 2020-09-02:
> lvgl v6.0

Please update to the `release/v6` branch, which contains 6.1 + any bugfixes after that. 6.0 is not supported.

zq190kg, 2020-09-02:
> > lvgl v6.0
> 
> Please update to the `release/v6` branch, which contains 6.1 + any bugfixes after that. 6.0 is not supported.

ok,tks