lvgl/lv_lib_png #5: can't display the png img in vs2019 monitor

banshanju, 2020-11-06:
I use LVGL's file system API, it work fine when i transform the img to bin file，but when i open png file directly，it display “nodata”.

I Initalize the decocer with lv_png_init() and add #define LV_PNG_USE_LV_FILESYSTEM 1.

It printf: ERROR 48:empty input buffer given to decoder. Maybe caused by non-existing file?

kisvegabor, 2020-11-08:
Hi,

Have you tried to open this example file? `lv_img_set_src(img, "P:lv_lib_lodepng/png_decoder_test.png");`

If so have you placed `lv_lib_lodepng` to the root folder of the project?

Can you "manually" open the file with `lv_fs_open`?

banshanju, 2020-11-09:
Yes, I placed lv_lib_lodepng to the root folder of the project, and try the example.
I will try to manually open the file with lv_fs_open, thank you!