lvgl/lv_lib_png #7: Lv_lib_png showing white rectangle

matinrn, 2021-01-21:
Hello 
I'm using esp32(arduino framework) + lvgl + lv_lib_lodepng.
when I try to show a png image it doesn’t work. it only shows a white rectangle with a “No data” text on the top left corner. I tried to Use PNG images from flash and file as instructed in the README file of the library and none of them works. the problem is only with png files because when i use a true color image it works fine.
I also Enabled the log module on LV_LOG_LEVEL_WARN level and the following lines are printed in the console:
`.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_img_cache.c@131->fnn: _lv_img_cache_open ->Image draw cannot open the image resource
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_draw_img.c@87->fnn: lv_draw_img ->Image draw error`
also the same code works in vs simulator and the problem only occurs when I’m using esp32.
**Code to reproduce**
`void setup()
{
  display_init();
  touch_init();
  lv_fs_if_init();
  LV_IMG_DECLARE(png_decoder_test);
  lv_obj_t *img = lv_img_create(lv_scr_act(), NULL);
  lv_img_set_src(img, &png_decoder_test);
}
void loop()
{
  lv_task_handler(); 
  delay(5);
}`
the lv_png_init() is inside the display_init() function.
**Screenshot and/or video**
[image]

embeddedt, 2021-01-21:
My educated guess would be that you are running out of heap space.

matinrn, 2021-01-30:
> 
> 
> My educated guess would be that you are running out of heap space.

I don't think that's the case here, because esp32 has 294k free heap before trying to show the image while the image size is only 20k.

kisvegabor, 2021-02-01:
What is the resolution of the image?

It needs minimum `hor. res. x ver. res. x 4 bytes`

matinrn, 2021-02-02:
> 
> 
> What is the resolution of the image?
> 
> It needs minimum `hor. res. x ver. res. x 4 bytes`

I'm using the image that came with the library (png_decoder_test.c) which is 200*150 and it needs 200*150*4=120kB which is about 1/3 of esp's free heap (294kB) in this project.

kisvegabor, 2021-02-03:
Do you see anything suspicious if you enable logging in lv_conf.h?

matinrn, 2021-02-04:
> 
> 
> Do you see anything suspicious if you enable logging in lv_conf.h?

I can't figure out the problem from the logs so I just post them here.

there's no log on LV_LOG_LEVEL_ERROR.
here's the log on LV_LOG_LEVEL_WARN:
`.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_img_cache.c@135->fnn: _lv_img_cache_open ->Image draw cannot open the image resource
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_draw_img.c@88->fnn: lv_draw_img ->Image draw error
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_img_cache.c@135->fnn: _lv_img_cache_open ->Image draw cannot open the image resource
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_draw_img.c@88->fnn: lv_draw_img ->Image draw error
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_img_cache.c@135->fnn: _lv_img_cache_open ->Image draw cannot open the image resource
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_draw_img.c@88->fnn: lv_draw_img ->Image draw error
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_img_cache.c@135->fnn: _lv_img_cache_open ->Image draw cannot open the image resource
.pio\libdeps\esp32dev\lvgl\src\lv_draw\lv_draw_img.c@88->fnn: lv_draw_img ->Image draw error`

and [here's](https://pastebin.com/9r0w64sE) the pastebin link for the log on LV_LOG_LEVEL_INFO.

kisvegabor, 2021-02-09:
Have you called `lv_png_init()`?

matinrn, 2021-02-10:
> 
> 
> Have you called `lv_png_init()`?

yes I have.

kisvegabor, 2021-02-11:
I've tested it on PC and worked well for me.

I suggest debugging the `devoder_open()` function to see why it can't open the imaga.

matinrn, 2021-02-11:
> I suggest debugging the `devoder_open()` function to see why it can't open the imaga.

I did it and function `lodepng_decode32()` which is called inside `decoder_open()` returns error **83** which means "memory allocation failed".
I used smaller pictures and it worked fine with a 150x100 png c array, but fails with a 200x100 one.
It's still confusing since esp32 has about 300kB free ram available and a 200x100 png c array needs only ~80kB of ram.

kisvegabor, 2021-02-12:
It's more than 80kB because the one decoder uses some internal working buffers too, but it shouldn't be an extra 220 kB.

Maybe it's somehow specific to ESP32 memory manager. 

You can try to change lodepng's malloc/free to lv_mem_alloc/realloc/free and give areally large LV_MEM_SIZE to LVGL.