lvgl/lv_lib_png #8: Does APNG support?

guoweilkd, 2021-02-05:


embeddedt, 2021-02-05:
I don't think lodepng (the underlying decoder being used) supports APNG. If you want animations I would suggest looking into https://github.com/lvgl/lv_lib_gif.

embeddedt, 2021-02-05:
It looks like APNG is an extension of PNG, so the first frame will probably display fine. However, there is no mechanism I know of to get lodepng to display the next frame.

closed 2021-02-05 by embeddedt

guoweilkd, 2021-02-22:
I use libpng+apng patch to implement APNG in LVGL. https://github.com/guoweilkd/lv_lib_apng
How to add it to lvgl Repo.

kisvegabor, 2021-02-22:
Happy to hear that you could make it work!

In this repo are using `lodepng` as png lib but it seems you are using a different one. 

Could you send a PR with the changes? We can merge it into a new branch and if it's better than the current `lodepng` lib we can replace it with yours.