lvgl/lv_lib_png #9: Load PNG from variable

beckmx, 2021-02-13:
Hello, guys, I have a doubt, if I follow the instructions to load data from the .c file, it loads, then if I provide the file directly with bytes, it shows me the "No data" message but in the log it says, it recognizes the file type, this is the snipet:

```
uint8_t tmp_img[20869];
File f = SPIFFS.open("/test.png", "r");
          Serial.print("Size:");
          Serial.println(f.size());
          if (f) {
            Serial.print("Bytes read:");
            Serial.println(f.readBytes((char *)tmp_img, f.size()));
            f.close();
          }
          const lv_img_header_t png_decoder_test_map_header2 = {
          LV_IMG_CF_RAW_ALPHA,
          0,
          2,
          200,
          150
          };
          const lv_img_dsc_t png_decoder_test2 = {
          png_decoder_test_map_header2,
          20869,
          tmp_img
          };
          lv_img_set_src(img1, &png_decoder_test2);
```

It does provide the bytes from what I can see and the logs say:

```
Size:20869
Bytes read:20869
Info: lv_img_set_src:  `LV_IMG_SRC_VARIABLE` type found 	(lv_img.c #180 lv_img_set_src())
Data x48
Data y118
Info: image draw: cache miss, close and reuse an entry 	(lv_img_cache.c #120 _lv_img_cache_open())
Warn: Image draw cannot open the image resource 	(lv_img_cache.c #135 _lv_img_cache_open())
Warn: Image draw error 	(lv_draw_img.c #88 lv_draw_img())
```

So, right afer it extracts the bytes it just goes directly to "No data"

kisvegabor, 2021-02-15:
The only I can imagine `tmp_img` is a local variable and goes out of the scope (hence destroyed).

What if you make it `static uint8_t tmp_img[20869];`?

beckmx, 2021-02-16:
Well I tried also, still getting the _no data_, that variable was defined at the top of my file, noton a scoped structure though

beckmx, 2021-02-16:
Ok, I am going to close it, I could load the image from flash which is an huge step forward

closed 2021-02-16 by beckmx

srall123, 2025-01-19:
Hello, how to figure it out. I have the same issue. Thx