lvgl/lv_lib_png #10: Problem when using canvas on top of png

beckmx, 2021-02-16:
Hello guys,

I noticed something weird, if I load my png alone (only the image on the screen) it loads fast (esp32 with 4mgs external ram) but if I add a canvas on top, it just gets to run very slow, I understand the canvas is this transparent piece of component that can be used to draw "primitives" to screen, but I am finding it quite difficult to make it faster with the canvas, can I draw a square on top of the PNG image without using the canvas?

embeddedt, 2021-02-17:
Do you get the same performance loss if you create a basic `lv_obj` and position it on top of the image?

beckmx, 2021-02-17:
If I use a basic object, it does render faster like this one:
```
lv_obj_t * obj1;
    obj1 = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_size(obj1, 100, 50);
    lv_obj_align(obj1, NULL, LV_ALIGN_CENTER, -60, -30);
```

The problem comes again, if I use another PNG image (extracted directly from flash as static) like this:

```
lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_scr_act(), NULL);
  lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_RELEASED, &icon_dinroom);
  lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_PRESSED, &icon_dinroom);
  lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_CHECKED_RELEASED, &icon_dinroom);
  lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_CHECKED_PRESSED, &icon_dinroom);
  lv_imgbtn_set_checkable(imgbtn1, true);
  // lv_obj_add_style(imgbtn1, LV_IMGBTN_PART_MAIN, &style);
  lv_obj_align(imgbtn1, NULL, LV_ALIGN_CENTER, 0, -40);

  /*Create a label on the Image button*/
  lv_obj_t * label = lv_label_create(imgbtn1, NULL);
  lv_label_set_text(label, "Button");
```

it kinda stops on the places where two PNG images are one on top of each other

kisvegabor, 2021-02-19:
Try `LV_IMG_CACHE_DEF_SIZE 2` in `lv_conf.h`.

beckmx, 2021-05-26:
I decided to change the image to jpg instead, it was a little bit of slow

closed 2021-05-26 by beckmx