lvgl/lv_lib_png #13: bug with LODEPNG_COMPILE_ALLOCATORS option

alsaleem00, 2021-03-19:
I am testing (lv_lib_png) performance and heap allocation.
I implemented my allocator (tracing/logging mallocs) to test software (basically the lodepng library) and enabling the MEM_BLOCK.

I found there is a missing FreeMem call which will result in memory leak [(here)](https://github.com/lvgl/lv_lib_png/blob/0d0bee85142a98b1dc905e43ab26bc72e1e85e1c/lodepng.c#L107)

I know is not lvgl code, so i went to lodepng site to report issue.
I found that this code is no longer exist.

In fact, I found that lv_lib_png is using "20180611" release. lodepng is now "20201017"

Regards,

kisvegabor, 2021-03-24:
Thanks for the report!

> In fact, I found that lv_lib_png is using "20180611" release. lodepng is now "20201017"

Would you be interested in sending a PR with the latest version?
I think the most important  change was added in this commit: https://github.com/lvgl/lv_lib_png/commit/66b664250720a8bcdd60e7b742d2243329e0e69b

alsaleem00, 2021-03-24:
I found that lodepng consumes (194KB, including the 37KB file size and 64KB image buffer). The file was only 128x128 pixels. 
That was a lot of memory for ESP32. I used pngle to decode png which consumes less,
However, pngle failed when i allocated memory from SPIRAM, while lodepng did not.

The new lodepng changed the code so that user can use his own memory allocation.

Let me check to evaluate the amount of work needed to use new lodepng in lv_lib_png. I will follow the same approach you used for the old one.

embeddedt, 2021-03-24:
We have also seen strangely high amounts of memory being used by lodepng in the MicroPython simulator so this seems to correlate with that. It would be great if a newer version of lodepng solved the issue.

(cc @amirgon)

alsaleem00, 2021-03-24:
Here is a log of memory allocation activities by lodepng (attached), both 2018 and 2020 versions.
There is some reduction in memory usage but still high.

[lodepng_mem_2018.txt](https://github.com/lvgl/lv_lib_png/files/6200261/lodepng_mem_2018.txt)
[lodepng_mem_2020.txt](https://github.com/lvgl/lv_lib_png/files/6200262/lodepng_mem_2020.txt)

alsaleem00, 2021-03-29:
lv_lib_png is now using lodepng newest release. So, this thread should be closed..

closed 2021-03-29 by alsaleem00