lvgl/lv_lib_png #15: how to enhance decoder's performance ?

shukewt, 2021-03-31:
If I show two 720p png,the decoder performance is terrible,the fps almost half lower than just show one picture,but if I replace one of it with a c array it works well.I'm sure the chip I'm using is powerful enouth to decode this so I wonder if there is some problem such as memory allowcation or some options I didn't notice? Can you give some advice about how to locate this problem.

shukewt, 2021-03-31:
One of the picture have to be refreshed periodically and the other one will only be loaded once,does the static one still keep the resource after it has been loaded?

kisvegabor, 2021-03-31:
Set `LV_IMG_CACHE_DEF_SIZE` to at least `2` for 2 images. It tells LVGL how many images to keep open. If it's 0 the image is decoded many times on demand instead of being cached.