lvgl/lv_lib_png #16: PNG test on lv_port_stm32f429_disco

WiktorBAbove, 2021-05-06:
Hello
Im trying to load the basic example on a stm32f429 discovery board.
Im not doing anything unusual,

```
    lv_init();
    tft_init();
    lv_png_init();
    .
    .
    .

    LV_IMG_DECLARE(png_decoder_test);
    lv_obj_t * imgpng = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(imgpng, &png_decoder_test);
    lv_obj_align(imgpng, NULL, LV_ALIGN_CENTER, 0, 0);
```

The only thing I get on the screen is: 
```
No
data
```

I just have one image, but I've #define LV_IMG_CACHE_DEF_SIZE       2
Is there anything else I need to think about?
Im using "7faf20592598e7f891cf2aceca32bbe6ff5fc36c"

WiktorBAbove, 2021-05-07:
That worked well!
I have tried to convert myown png to 50x50 like `png_decoder_test`.
```
const lv_img_dsc_t png_decoder_test = {
  .header.always_zero = 0,
  .header.w = 50,
  .header.h = 50,
  .data_size = 5158,
  .header.cf = LV_IMG_CF_RAW_ALPHA,
  .data = png_decoder_test_map,
};
```
and mine is:
```
const lv_img_dsc_t testtubeslab2 = {
  .header.always_zero = 0,
  .header.w = 50,
  .header.h = 50,
  .data_size = 3727,
  .header.cf = LV_IMG_CF_RAW_ALPHA,
  .data = testtubeslab2_map,
};
```
It is smaller in size however it is showing just `PNG` on the screen.
I converted it by:
` php ../lv_utils/img_conv_core.php "name=testtubeslab2&img=./png/testtubeslab2.png&format=c_array&cf=raw_alpha"`
I am using `php7.4-gd` and `php7.4-cli` if that matters?
Am i missing something crucial for it to work?

kisvegabor, 2021-05-10:
Could you send the PNG image you used?

WiktorBAbove, 2021-05-10:
I think i figured out what was wrong!
1. resized the original with windows10's builtin converter (powertoys)
2. resized with imagick (convert)
What I got was interesting:
1) 3071bytes
2) 3727bytes

Both files can be opened and viewed. However the current decoder cant handle an image that is resized with Imagick.
`convert orig.png -resize 50x50 resized.png`

Now I've actually managed to combine, a transparent PNG together with a 320x240 transparent GIF animation, without graphical issues.

I suppose we can close this issue, thank you for quick response! and your work!

embeddedt, 2021-05-10:
I'd be curious to know if the same issue happens when converting the PNG to LVGL format using the experimental [JS converter](https://embeddedt.github.io/lv_img_conv/). There are many variants of PNG and the PHP converter tends to fail on several of them.