lvgl/lv_lib_png #19: PNG photo 8 bitcolor someting wrong

Adalaider, 2021-07-21:
maybe it should be：
#elif LV_COLOR_DEPTH == 8
    lv_color32_t * img_argb = (lv_color32_t*)img;
       lv_color_t c;
       uint32_t i;
       for(i = 0; i < px_cnt; i++) {
           c = lv_color_make(img_argb[i].ch.blue, img_argb[i].ch.green, img_argb[i].ch.red);
           img[i*2 + 1] = img_argb[i].ch.alpha;
           img[i*2 + 0] = c.full;
       }

Adalaider, 2021-07-21:
I‘m green hand， I found that 8-bit PNG can be displayed in this way, but I don't know what risks there are, so I hope you can update this PNG library ，thank you so much

kisvegabor, 2021-07-21:
Thanks, I fixed it here as you suggested: https://github.com/lvgl/lv_lib_png/commit/2bb254cbe545568c3186138cd773d7413b0d3cc5

closed 2021-07-21 by kisvegabor