lvgl/lv_lib_png #22: error: conflicting types for 'decoder_info'

TheAnton205, 2021-09-17:
Attempted to build using pio and I run into this error:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stack-core2.html
PLATFORM: Espressif 32 (3.2.1) > M5Stack Core2
HARDWARE: ESP32 240MHz, 6.25MB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40201.210513 (4.2.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\core2foraws\components\lv_lib_png\lv_png.o
components\lv_lib_png\lv_png.c:31:37: warning: 'struct _lv_img_decoder_t' declared inside parameter list will not be visible outside of this definition or declaration
 static lv_res_t decoder_info(struct _lv_img_decoder_t * decoder, const void * src, lv_img_header_t * header);
                                     ^~~~~~~~~~~~~~~~~
components\lv_lib_png\lv_png.c: In function 'lv_png_init':
components\lv_lib_png\lv_png.c:54:37: warning: passing argument 2 of 'lv_img_decoder_set_info_cb' from incompatible pointer type [-Wincompatible-pointer-types]
     lv_img_decoder_set_info_cb(dec, decoder_info);
                                     ^~~~~~~~~~~~
In file included from components\core2forAWS\tft\lvgl/lvgl/src/lv_core/../lv_draw/lv_draw_img.h:16,
                 from components\core2forAWS\tft\lvgl/lvgl/src/lv_core/lv_obj.h:29,
                 from components\core2forAWS\tft\lvgl/lvgl/lvgl.h:32,
                 from components\lv_lib_png\lv_png.c:12:
components\core2forAWS\tft\lvgl/lvgl/src/lv_core/../lv_draw/lv_img_decoder.h:201:85: note: expected 'lv_img_decoder_info_f_t' {aka 'unsigned char (*)(struct _lv_img_decoder *, const void *, struct <anonymous> *)'} but argument is of type 'lv_res_t (*)(struct _lv_img_decoder_t *, const void *, lv_img_header_t *)' {aka 'unsigned char (*)(struct _lv_img_decoder_t *, const void *, struct <anonymous> *)'}
 void lv_img_decoder_set_info_cb(lv_img_decoder_t * decoder, lv_img_decoder_info_f_t info_cb);
                                                             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
components\lv_lib_png\lv_png.c: At top level:
components\lv_lib_png\lv_png.c:70:37: warning: 'struct _lv_img_decoder_t' declared inside parameter list will not be visible outside of this definition or declaration
 static lv_res_t decoder_info(struct _lv_img_decoder_t * decoder, const void * src, lv_img_header_t * header)
                                     ^~~~~~~~~~~~~~~~~
components\lv_lib_png\lv_png.c:70:17: error: conflicting types for 'decoder_info'
 static lv_res_t decoder_info(struct _lv_img_decoder_t * decoder, const void * src, lv_img_header_t * header)
                 ^~~~~~~~~~~~
components\lv_lib_png\lv_png.c:31:17: note: previous declaration of 'decoder_info' was here
 static lv_res_t decoder_info(struct _lv_img_decoder_t * decoder, const void * src, lv_img_header_t * header);
                 ^~~~~~~~~~~~
components\lv_lib_png\lv_png.c:31:17: warning: 'decoder_info' used but never defined
components\lv_lib_png\lv_png.c:70:17: warning: 'decoder_info' defined but not used [-Wunused-function]
 static lv_res_t decoder_info(struct _lv_img_decoder_t * decoder, const void * src, lv_img_header_t * header)
                 ^~~~~~~~~~~~
Compiling .pio\build\core2foraws\lwip\lwip\src\core\pbuf.o
Compiling .pio\build\core2foraws\lwip\lwip\src\core\raw.o
Compiling .pio\build\core2foraws\lwip\lwip\src\core\stats.o
Compiling .pio\build\core2foraws\lwip\lwip\src\core\sys.o
Compiling .pio\build\core2foraws\lwip\lwip\src\core\tcp.o
Compiling .pio\build\core2foraws\lwip\lwip\src\core\tcp_in.o
*** [.pio\build\core2foraws\components\lv_lib_png\lv_png.o] Error 1
```

TheAnton205, 2021-09-17:
Additional error message:
```
declaration is incompatible with "lv_res_t decoder_info(struct _lv_img_decoder_t *decoder, const void *src, lv_img_header_t *header)" (declared at line 70)
```

and at line 70:
```
expected an identifier
```

TheAnton205, 2021-09-17:
Additional information:

```
declaration is incompatible with "lv_res_t decoder_info(struct _lv_img_decoder_t *decoder, const void *src, lv_img_header_t *header)" (declared at line 69)
```

kisvegabor, 2021-09-17:
Do you use lvgl v7?
 If so please use the release/v7 branch of this lib.

TheAnton205, 2021-09-17:
> Do you use lvgl v7?
> If so please use the release/v7 branch of this lib.

I have the latest version of this library if this is what you mean. I also attempted to use a previous version to no avail.

TheAnton205, 2021-09-17:
If the latest version is v7, then yes I have it. Additionally, I attempted to use the version from June, yet to no avail.

kisvegabor, 2021-09-17:
> Additionally, I attempted to use the version from June, yet to no avail.

"June" version of LVGL or lv_lib_png? Have you tried with the release/v8 branc of lv_lib_png as I suggested above.

leverfor, 2022-05-10:
Bro,I'm solving this problem, too.Have you solved it？

kisvegabor, 2022-11-29:
@tobiasraak 
 Which LVGL version do you use? And how do you use the PNG image?

tobiasraak, 2022-11-29:
@kisvegabor I use the latest version [v8.3.3](https://github.com/lvgl/lvgl/releases/tag/v8.3.3). I store the PNG on an esp32 flash storage. But it is not possible for me to access this png. In fact, all settings for a filesystem in the lv_conf.h need a drive letter. Since I use SPIFFS I do not have a drive letter, I am not able to parse the png. Do you have any example for this scenario? I was not able to get it running.

tobiasraak, 2022-11-29:
@kisvegabor I really need help espicially with the drive letter. How do I create a drive letter with any of the supported filesystems that are listetd in the lv_conf.h? I can assign a drive number with FATFS for instance but there is no way (and I could not find any example) on how to create a filesystem image on the esp32 with a drive letter. Could you please provide some help to me with that? I would really appreciate this.

kisvegabor, 2022-12-01:
For ESP all you need  to do is enabling `LV_USE_FS_STDIO` with any letter, e.g. `A`. After that you need to prefix the SPIFFS with `A:`.

E.g. `A:/spiffs/tmp/myfile.png`.