lvgl/lv_lib_png #23: invalid ADLER32 while frawing image

glory-man, 2021-09-28:
I try to draw attached png on the screen and received error 58 - invalid ADLER32 encountered. At the same time, this image is normally opened by any desktop-image-viewer. Is this an image problem? Is it possible to fix it?
[image]

kisvegabor, 2021-09-29:
Maybe the image is slightly corrupted and full featured desktop viewers can handle it. 

I suggest opening and saving the image. Hopefully it'll fixes it.

glory-man, 2021-09-29:
> Maybe the image is slightly corrupted and full featured desktop viewers can handle it.
> 
> I suggest opening and saving the image. Hopefully it'll fixes it.

I thought about it, but unfortunaly not helped. I was able to draw image by disabling adler32 checking.

glory-man, 2021-09-29:
By the way errors 57 & 58 description tells that - (checking CRC can be disabled) / (checking ADLER32 can be disabled). So in lodepng.h may be added code like this
`#ifdef LV_PNG_IGNORE_ADLER32
#define LODEPNG_IGNORE_ADLER32 1
#else
#define LODEPNG_IGNORE_ADLER32 0
#endif

#ifdef LV_PNG_IGNORE_CRC
#define LODEPNG_IGNORE_CRC 1
#else
#define LODEPNG_IGNORE_CRC 0
#endif
`
So in decompress initialization function may be used LODEPNG_IGNORE_CRC/LODEPNG_IGNORE_ADLER32 instead of 0/1. In this case we don't need edit sources to configure library.

kisvegabor, 2021-09-30:
I've just downloaded the image form your first post and it worked for me. 

lodepng was update a couple of month ago. Do you use the latest version from the master branch?

glory-man, 2021-09-30:
> What if you overwrite `lodepng.c/h` with the latest versions? There shouldn't be v8 related stuff there.
> 
> It might also worth a try to download the image from your post. Maybe GitHub converted it.

I'll try it. Thanks.

glory-man, 2021-10-01:
> What if you overwrite `lodepng.c/h` with the latest versions? There shouldn't be v8 related stuff there.
> 
> It might also worth a try to download the image from your post. Maybe GitHub converted it.

I tried latest version of lodepng.c. This file uses
`lv_fs_seek(&f, 0, SEEK_END)`
with 3 params (inside lodepng_filesize), while in lvgl_v7 it has only 2 params. So I use 
`lv_fs_size(&f, &size);`
to detect size of file.
But file still not opened without disabling adler32 checking.

kisvegabor, 2021-10-04:
And have you downloaded the image from GitHub (from your first comment)?

glory-man, 2021-11-02:
> And have you downloaded the image from GitHub (from your first comment)?

Yes. I tried with image downloaded from github.
In debugger 
`unsigned ADLER32 = lodepng_read32bitInt(&in[insize - 4]);`
called in _lodepng_zlib_decompressv_ function gives me 0xC0C64FFD, while
`unsigned checksum = adler32(out->data, (unsigned)(out->size));`
gives 0xBF592DC8, out->size = 0x230

kisvegabor, 2021-11-05:
I have no idea :slightly_frowning_face: 

I've tested it again and `out->size = 0x230` for me too but `checksum = 0xc0c64ffd`.

glory-man, 2021-11-05:
> I have no idea 🙁
> 
> I've tested it again and `out->size = 0x230` for me too but `checksum = 0xc0c64ffd`.

I founded a problem. I use custom allocator function, where lodepng_realloc was incorrect. Fixed by using lv_mem_realloc.

kisvegabor, 2021-11-05:
Glad to hear that!

closed 2021-11-05 by kisvegabor