lvgl/lv_lib_png #26: 'LV_PNG_USE_LV_FILESYSTEM' is never used?

TaXue-DuBa, 2021-11-19:
Hello, I use the ` lv_lib_png ` have a problem.
First, I need to state steps before asking the question:
1.  I add the `lv_fs_if` in my project,  and I am sure that it is available. Because I can load my `qingtian.bin` used with `lv_img_set_src`.
2.  Then, I add the `#define LV_PNG_USE 1 #define LV_PNG_USE_LV_FILESYSTEM 1` at the end of 'lv_cong.h'.But Clion(platformio) remind `#define LV_PNG_USE 1 #define LV_PNG_USE_LV_FILESYSTEM 1 is never used.`
[image]
3. Create `lv_obj_img` in my project.[image]
4. Include and init it.
[image]
5. My monitor(ST7789) show `"No data"` in `weather_img`obj, It is wrong.

So, I don`t know what happend? Where is the problem?

kisvegabor, 2021-11-19:
Can you update to the latest lvgl version? A few weeks ago we have merged the png and file system library to lvgl to make them work better together.  See https://docs.lvgl.io/master/libs/png.html

TaXue-DuBa, 2021-11-21:
> Can you update to the latest lvgl version? A few weeks ago we have merged the png and file system library to lvgl to make them work better together. See https://docs.lvgl.io/master/libs/png.html

emm. I try LVGL8.1.0 on ESP32 with arduino, I init the `SPI.h and SD.h` in `fs_init()` function, but it warn me `[E][sd_diskio.cpp:123] sdSelectCard(): Select Failed"`.
So LVGL file system can not work in lvgl8.1.0 well. But LVGL8.0.1 work good(lv_lib_png can not work).
So, I think the logic changed possibly which is using `Fatfs` System?

kisvegabor, 2021-11-26:
In v8.1 you can enable the file system APIs too from lv_conf.h. So you don't need lv_fs_if.

TaXue-DuBa, 2021-11-28:
> 

Yes, I know. I meant I had used the file system APIs from lv_conf.h in LVGL8.1. But it not work.
lv_fs_if can work in LVGL8.0.2.

kisvegabor, 2021-11-30:
How do you mean "not work"? What happens exactly? 

Some questions to see what could be the problem:
- Can you just open a txt file read and close it with `lv_fs_...` API? If not please send a code to reproduce the issue. 
- How does the related part of `lv_conf.h` look like? 
- What platform/hardware do you use?

TaXue-DuBa, 2021-12-26:
Emmmm, That`s is my personal problem!!!!! I `m so sorry!!! I find i need `VSPI` but transfered `HSPI` which had used by LVGL...
Thank you answer again!!!

closed 2021-12-26 by TaXue-DuBa

bhupiister, 2022-01-02:
I am having this weird problem with SD card. I am using M5 Core 2 and lv_Arduino library for lvgl graphics. My SD card is initialized properly and i can read the files. The problem occurs when i read the list the files using lvgl more than couple of times. I can see that if i keep my touch enabled on any lvgl object for long time, it troubles with my SD card module and i get this error "sdSelectCard(): Select Failed". 
For example, if i keep my touch on any list for long time but do not select any option, and then try to access my Sd card. i get the above error.

One more clarification i need is, if i use 
`lv_obj_t *win = lv_win_create(lv_scr_act(), NULL)`
Will this create an issue? and on closing the window, is it automatically destroyed?

TaXue-DuBa, 2022-01-02:
> I am having this weird problem with SD card. I am using M5 Core 2 and lv_Arduino library for lvgl graphics. My SD card is initialized properly and i can read the files. The problem occurs when i [...]

The img show normally When you use the `<IMG_NAME>.bin` ?

Your No.2 question I do not know. But I recommand you to delete it manually.

bhupiister, 2022-01-02:
> The img show normally When you use the `<IMG_NAME>.bin` ?
>
> Your No.2 question I do not know. But I recommand you to delete it manually.

I read more about it. It seems the issue is related to SPI bus shared by SD card module and display on M5stack core 2.