lvgl/lv_lib_png #27: Png picture distortion

yin2021, 2021-12-22:
Hello, I use the lv_lib_png have a problem.
LVGL version: 7.0.
My PNG image is RGBA8888, and the LV_COLOR_DEPTH is set to 16 because the device only supports 16bit pix. when trying to display some image with gradient color, there is a picture distortion problem.
original image is:
[image]
in fact, displaying image is:
[image]
I find the problem is in function "convert_color_depth" in lv_png.c. In this function, RGB888 is converted to RGB 565, and low bits of R byte、G byte and B byte are dropped, which creates such a problem. 
So，there is a better way to deal with it. Thank you for your replying!

embeddedt, 2021-12-22:
This is a known limitation of 16-bit color. To work around it, you need to dither the image. Support for this is likely to be added to LVGL (v8) soon, but you may also be able to do it by manually dithering the image in GIMP or another tool first.

kisvegabor, 2021-12-22:
You can also use e.g. 10 px high dithered image and let LVGL tile it.

yin2021, 2021-12-23:
> This is a known limitation of 16-bit color. To work around it, you need to dither the image. Support for this is likely to be added to LVGL (v8) soon, but you may also be able to do it by manually dithering the image in GIMP or another tool first.

1. Thank you for your replying. I got a new image generated by GIMP, which's performance is better. However, when it is displayed in device, there are still some strip, but not on the PC simulator(also LV_COLOR_DEPTH 16). I don't know why?
2.  I think it is necessary to integrate image dithering in LVGL, because the device will display image download from cloud, which could not be manually dithered. 
3. By the way, when will it be released, and we are looking forward to it. If it will take a long time to release, can you tell me how to achieve it by myself. I've tried some methods, the following is relatively good
[image]
Is this method similar to yours

yin2021, 2021-12-23:
> You can also use e.g. 10 px high dithered image and let LVGL tile it.

I'm sorry I don't quite understand what you mean. I don't know what the number of pixels has to do with this problem. Can you explain. Thank you!

kisvegabor, 2022-01-04:
Let's say you can have a dithered 10x200 image and you can set it for a 200x200 `lv_img` widget. (LVGL automatically tiles the image source if it's smaller than the widget size.)